ln-12/blainder-range-scanner #1: Installation of the add-on

mabuh004, 2021-09-08:
I tried to install the add-on using the provided way, which is: 
(Copy the range_scanner folder to C:\Program Files\Blender Foundation\Blender 2.83\2.83\scripts\addons_contrib (Windows).)

However, when I try to activate the add on in Blender, this message appears,  
[image]

How it can be solved? 

Regards.

Reitmania, 2021-09-08:
Hi, looks like you don't have the pypng python package installed on your computer. Try `pip install pypng`. 

Best regards, reiti

ln-12, 2021-09-08:
The installation under Windows can be tricky. In case it does not work after the pip command @Reitmania mentioned, try to force the installation of all dependencies for Blenders included python with **admin privileges** (important): 
```
"C:\Program Files\Blender Foundation\Blender 2.83\2.83\python\bin\python.exe" -m pip install laspy h5py pascal_voc_writer pypng pyyaml
```

mabuh004, 2021-09-08:
how can I try these things? in python? or in blender?

Reitmania, 2021-09-08:
neither python nor blender, you should open a Windows command window and use the mentioned commands - at best the one of @ln-12 above

(e.g. if you install pypng with pip, you can check the installation in python via "import png". The bug should also disappear in the BlAInder add-on)

mabuh004, 2021-09-08:
I opened the command window and used the mentioned command, 
[image]

After that I checked the python using import png, but the bug is still existed in the add-on.

ln-12, 2021-09-08:
Did you run it as admin?

mabuh004, 2021-09-08:
@ln-12 when I run it as admin, this happens
[image]

ln-12, 2021-09-08:
Well, I tried the install procedure on a fresh install of Windows and your error only occurred when running the command prompt as normal user. After starting it as administrator ([for description see here](https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-server-2012-r2-and-2012/jj717276(v=ws.11))), it worked as intended. Your error message also suggests that you do not run it as admin.

mabuh004, 2021-09-14:
@ln-12 @Reitmania Thank you so much! 
It is working now.

closed 2021-09-14 by mabuh004