ln-12/blainder-range-scanner #2: Add requirements.txt for manual pip installation of dependencies

fkaufm, 2021-11-05:
Automatic search for dependencies did not work for me (tested Win10 and Ubuntu 20.04). When trying to install dependencies manually, I ran into an issue with laspy. By default, laspy2.x will be installed, but this is not compatible to las export in blainder. 
I would suggest to add a requirements file to declare required versions: [requirements.txt](https://github.com/ln-12/blainder-range-scanner/files/7487621/requirements.txt). 
Note: I tested the las export with laspy 1.7.0, no other tests so far.

ln-12, 2021-11-05:
Hey @fkaufm, thanks for your suggestion! I'll have a look at it in the next few days :)

ln-12, 2021-11-07:
Hey @fkaufm, thanks again for your contribution! I added your requirements.txt and enhanced it with some extra comments to use it in the automatic dependency installation. For me both manual and automatic installation work now. Feel free to add some info if you run into any trouble :)

fkaufm, 2021-11-09:
Hey @ln-12 thanks for your reply and quick fixes. I tried to install blainder today but constantly run into issues. When trying to activate the add-on, I constantly run into trouble: 
```
Exception in module register(): /home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/__init__.py
Traceback (most recent call last):
  File "/home/kaufmann/blender-2.93.4-linux-x64/2.93/scripts/modules/addon_utils.py", line 386, in enable
    mod.register()
  File "/home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/__init__.py", line 21, in register
    user_interface.register()
  File "/home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/ui/user_interface.py", line 2349, in register
    requirementsFile = open('requirements.txt', 'r')
FileNotFoundError: [Errno 2] No such file or directory: 'requirements.txt'
```
Where do we have to put the requirements.txt that it can be found from the script? I tried putting it into range_sanner/ and range_scanner/ui/, no success so far. 

I succeeded installing the dependencies manually via ` blender-2.93.4-linux-x64/2.93/python/bin/python3.9 -m pip install -r blainder-range-scanner/requirements.txt`, with `blender-2.93.4-linux-x64/2.93/python/bin/python3.9 -m pip freeze` output: 
```
h5py==3.5.0
Jinja2==3.0.2
laspy==2.0.3
MarkupSafe==2.0.1
numpy==1.21.4
pascal-voc-writer==0.1.4
pypng==0.0.21
PyYAML==6.0
```
but blainder gets stuck in the addon installation / import test loop:
[image]

 ```
Exception in module register(): /home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/__init__.py
Traceback (most recent call last): 
File "/home/kaufmann/blender-2.93.4-linux-x64/2.93/scripts/modules/addon_utils.py", line 386, in enable
    mod.register()
  File "/home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/__init__.py", line 21, in register
    user_interface.register()
  File "/home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/ui/user_interface.py", line 2341, in register
    bpy.utils.register_class(cls)
ValueError: register_class(...): already registered as a subclass 'WM_OT_INSTALL_DEPENDENCIES'

```
Unfortunately, I could not use the addon.
I tried it with Blender2.93 and Ubuntu 20.04, no other platforms so far. 
Again, some help on this will be appreciated.

fkaufm, 2021-11-09:
I now did a test on Win10 with Blender2.93. Addon and dependency installation work "silently", no issues to report.

ln-12, 2021-11-09:
@fkaufm the first error was my bad. I am working on an updated version which should find the file. I did only test it on Windows and in debug mode as I wanted to quickly push a fix.

The second error including `WM_OT_INSTALL_DEPENDENCIES` occurs when you try to install the add-on again. It can be fixed by removing the add-on and restarting Blender.

ln-12, 2021-11-09:
I tested the automatic installation again with Windows 11 and a fresh copy of Blender 2.93.5 and the add-on. Works fine for me now.

closed 2021-11-09 by ln-12

fkaufm, 2021-11-16:
I tested this on Ubuntu 20.02 LTS now. I came across two things: 
1) It only works with the .tar.xz archive from blender.org, not with snap package (default from Ubuntu Software). Snaps seem to be read-only file systems, so installing python dependencies will not work. Probably, there are ways to do but I am not aware of. 
2) There are constant conflicts between the system python interpreter and blender's python on Ubuntu (and probably most Linux Distros). I was able to fix this by setting `os.environ["PYTHONNOUSERSITE"] = "1"` and resetting environment variables after installing pip in `install_pip()` as you indicated for python module installation function. That makes it work on Ubuntu. @ln-12: Does this make sense to you? And is it worth opening a pull request?

ln-12, 2021-11-16:
1. That's a good point, I actually never tested the Blender snap package. For now, you have to go with the archive then unless I (or someone else) finds a way to do it with snap packages.
2. So you suggest to insert `os.environ["PYTHONNOUSERSITE"] = "1"` again [here](https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/ui/user_interface.py#L122)? I never had problems with that, but of course this could be added. And feel free to open a PR, I will do my best to test and merge it.

commit a3380cca (2021-11-07): Improved dependency installation (closes #2)