ln-12/blainder-range-scanner #6: Background operation or Context override raises AttributeError

kalbfled, 2021-11-27:
**Describe the bug**

The add-on crashes with Blender in [background mode](https://docs.blender.org/manual/en/latest/advanced/command_line/arguments.html#render-options) or when [overriding the context](https://docs.blender.org/api/current/bpy.ops.html#overriding-context).  Both use cases raise AttributeError when calling the scan_static function (and presumably others).

This precludes headless, batch processing of .blend files.

**To Reproduce**

For a script named script.py that calls scan_static, and for a .blend file with the range_scanner add-on enabled, run the script from the command line like this: `blender -b <blend file> -P script.py`

You should see a traceback like this:
```
Traceback (most recent call last):
  File <script.py>, line 5, in <module>
    range_scanner.ui.user_interface.scan_static(
  File ".../range_scanner/ui/user_interface.py", line 1621, in scan_static
    performScan(context, dependencies_installed, properties)
  File ".../range_scanner/ui/user_interface.py", line 1387, in performScan
    modifyAndScan(context, dependencies_installed, properties, None)
  File ".../range_scanner/ui/user_interface.py", line 1284, in modifyAndScan
    generic.startScan(context, dependencies_installed, properties, objectName)
  File ".../range_scanner/scanners/generic.py", line 305, in startScan
    mode = bpy.context.area.type
AttributeError: 'NoneType' object has no attribute 'type'
```

Trying to avoid that problem by using a context override yields the following traceback:
```
Traceback (most recent call last):
  File <script.py>, line 29, in <module>
    range_scanner.ui.user_interface.scan_static(
  File ".../range_scanner/ui/user_interface.py", line 1564, in scan_static
    scene = context.scene
AttributeError: 'dict' object has no attribute 'scene'
```
The problem here is that the code expects a bpy_types.Context instance, but an overridden context is a dictionary.

**Expected behavior**

The add-ons in Blender's core all support overriding the context.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Blender version: 2.93.6

kalbfled, 2021-11-27:
Calls to scan_sonar do not seem to have this problem.

ln-12, 2021-11-30:
Thanks for reporting, this indeed seems to be broken. I will have a look as soon as I can :)

ln-12, 2021-12-06:
I removed the context/area usage for now as it seems to be only needed for my internal development. Could you please check out https://github.com/ln-12/blainder-range-scanner/pull/9 to test it? Thanks!

kalbfled, 2021-12-11:
Thank you for making the changes.  I'm using your add-on for work, and the initial focus was on the sonar scan.  I think that we'll proceed to the lidar scans before too long.  When that happens, I will test and let you know the outcome.

ln-12, 2021-12-11:
Sure, if you need any further assistance (or a commercial license for closed source changes) feel free to ask :)

ln-12, 2022-05-13:
Hey @kalbfled, do you need further assistance or can this issue be closed?

kalbfled, 2022-05-13:
I'm good.  Thank you for your responsiveness.

closed 2022-05-13 by kalbfled