ln-12/blainder-range-scanner #7: CSV exporter writes undesired blank spaces

kalbfled, 2021-12-04:
**Describe the bug**
When I run a sonar scan on sonar_example.blend and output to CSV, the rows contain a bunch of blank spaces.  This breaks trying to read the CSV with [numpy.loadtxt](https://numpy.org/doc/stable/reference/generated/numpy.loadtxt.html).  I might be able to work around this by passing the `converters` argument to loadtxt.

'''
...$ head -n 2 sonar_example_frame_300.csv 
c a t e g o r y I D ; p a r t I D ; X ; Y ; Z ; d i s t a n c e ; X _ n o i s e ; Y _ n o i s e ; Z _ n o i s e ; d i s t a n c e _ n o i s e ; i n t e n s i t y ; r e d ; g r e e n ; b l u e ;
0 ; 0 ; 1 . 5 7 0 ; 4 . 9 4 7 ; 0 . 0 6 9 ; 2 . 8 7 0 ; 1 . 5 7 0 ; 4 . 9 4 7 ; 0 . 0 6 9 ; 2 . 8 7 0 ; 0 . 6 4 7 ; 0 . 8 0 0 ; 0 . 5 0 8 ; 0 . 0 7 7
'''

**To Reproduce**

Run a sonar scan on sonar_example.blend and output to CSV,

**Expected behavior**

no unnecessary spaces

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Blender version: 3.0.0

**Additional context**

I think the problem is with the "%.3" format notation in the CSV exporters code.

ln-12, 2021-12-05:
Hey @kalbfled, thanks for reporting! This was indeed not as intended. I created a pull request [here](https://github.com/ln-12/blainder-range-scanner/pull/8), could you please have a look, if that commits fixes the issue for you?

kalbfled, 2021-12-05:
Yes, it seems to resolve the issue.  Thank you for the quick turnaround.

closed 2021-12-06 by ln-12