ln-12/blainder-range-scanner #10: Blainder crashing when modifiers present in scene

fkaufm, 2022-01-04:
In our research we perform some blainder scannings of IFC files imported with Blender BIM. In some cases, blainder crashes when hitting "Generate scans" button not performing scans. This seems to be related to modifiers being present in the scene. 

Could you please clarify it this is an actual bug or whether this behaviour is expected and modifiers should thus not be used for scanning with blainder?

**Steps to reproduce the behavior:**
1. Open minimum example file: https://seafile.rlp.net/d/f2a5db7f0fa043daafc0/
2. Set path and file name for point cloud
3. Hit 'Generate point cloud'
4. See error

**Expected behavior**
Blainder performing scan according to configuration and saving output files to specified path. 

**Error**

location: <unknown location>:-1
Error: Python: Traceback (most recent call last):
  File "/home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/ui/user_interface.py", line 1637, in execute
    performScan(context, dependencies_installed, properties)
  File "/home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/ui/user_interface.py", line 1387, in performScan
    modifyAndScan(context, dependencies_installed, properties, None)
  File "/home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/ui/user_interface.py", line 1284, in modifyAndScan
    generic.startScan(context, dependencies_installed, properties, objectName)
  File "/home/kaufmann/.config/blender/2.93/scripts/addons/range_scanner/scanners/generic.py", line 228, in startScan
    bpy.ops.object.modifier_apply(apply_as='DATA', modifier=modifier.name)
  File "/home/kaufmann/blender-2.93.5-linux-x64/2.93/scripts/modules/bpy/ops.py", line 132, in __call__
    ret = _op_call(self.idname_py(), None, kw)
TypeError: Converting py args to operator properties: : keyword "apply_as" unrecognized

location: <unknown location>:-1

**Environment:**
 - OS: Ubuntu 20.04 LTS
 - Blender 2.93 LTS

ln-12, 2022-01-04:
Thanks for reporting, that does not seem right. Blainder applies all modifiers before running the actual simulation to get the full geometry (for example with particle systems). I will have a closer look in the next few days. In the meantime you could try to remove modifiers you don't really need or try to apply them manually and then comment the lines [here](https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/scanners/generic.py#L227-L228).

ln-12, 2022-01-05:
It should be fixed in commit 3827722c6718bf5dbe114a1155e4ea2919506b35, can you give it a try?

ln-12, 2022-01-05:
I just noticed that yout `IfcOpeningElement/Teilelement:2388325:1` elements have a texture set and are therefore used to reflect the rays. I dont' know if you are using a window texture, but if this opening should only be a cutout in the wall, you need to remove the texture to skip this it while scanning.

ln-12, 2022-05-13:
Hey @fkaufm, do you need any further help or can we close this issue?

fkaufm, 2022-05-19:
Let's close this issue, thanks a lot for your help!