ln-12/blainder-range-scanner #13: Scan Parameters

thomasTUK, 2022-03-18:
Hello, I was not able to post it as pull request....
Great work so far!
I would like to have a uneven vertical FOV, for example -15° and 25°.
Also would like to save my own presets as scanner.

Many thanks!

ln-12, 2022-03-20:
Hey, thanks for your suggestion. I don't think that this needs to be implemented separately. Why don't you just position you camera to +5° and scan -20° to +20°? That would be the easiest approach without any changes in code. 

To save a scanner preset, you can just add your values to your local [presets.yaml](https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/ui/presets.yaml).

thomasTUK, 2022-03-20:
Ah, thanks! I will try both suggestions!

ln-12, 2022-05-13:
Do you need any further help or can we close this issue? :)

closed 2022-06-20 by ln-12