ln-12/blainder-range-scanner #19: AttributeError: 'Scene' object has no attribute 'scannerProperties'

ozge-albayrak, 2022-08-02:
Hi!
I am trying to use your toolbox for my application and I would like to utilize your scripts. In order to do that I have tried to run the script_usage.blend file. However, I am getting the following error.

``  File "C:\Program Files\Blender Foundation\Blender 2.93\2.93\scripts\addons_contrib\range_scanner\ui\user_interface.py", line 1565, in scan_static
    properties = scene.scannerProperties
AttributeError: 'Scene' object has no attribute 'scannerProperties' ``

Since I am pretty new to Blender scripting I couldn't quite figure out the issue. 
I would appreciate if someone can guide me in the right direction.
Thanks!

ln-12, 2022-08-03:
Hey @ozge-albayrak, could you please have a look under `Edit` > `Preferences` > `Add-ons`? The checkbox for `3D View: range_scanner` needs to be checked, otherwise the add-on is not installed properly.

[image]

ozge-albayrak, 2022-08-04:
Hi! 

Thanks for the reply! I am trying to use scripting only rather than GUI. Thus, I have followed the steps given for only "Script Usage". Apparently, you don't need to copy paste the range_scanner folder into Blender 2.93/2.93/scripts/addons_contrib folder. Once I removed that folder from addons_contrib and checked the checkbox of 3D View: range_scanner, I think the add on was installed properly. 

Now, when I run the script_usage.blend file, the rendering starts, however, the script fails and I get this error. 

```
Scanning scene: [--------------------] 0.0%Traceback (most recent call last):
  File "C:\Users\oezge\PATH\blender\blainder-range-scanner\example_scenes\script_usage.blend\Text", line 61, in <module>
  File "C:\Users\oezge\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\range_scanner\ui\user_interface.py", line 1539, in scan_sonar
    performScan(context, dependencies_installed, properties)
  File "C:\Users\oezge\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\range_scanner\ui\user_interface.py", line 1387, in performScan
    modifyAndScan(context, dependencies_installed, properties, None)
  File "C:\Users\oezge\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\range_scanner\ui\user_interface.py", line 1284, in modifyAndScan
    generic.startScan(context, dependencies_installed, properties, objectName)
  File "C:\Users\oezge\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\range_scanner\scanners\generic.py", line 271, in startScan
    sonar.performScan(context,
  File "C:\Users\oezge\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\range_scanner\scanners\sonar.py", line 336, in performScan
    partIDIndex = closestHit.target.material_slots[materialMappings[closestHit.target][closestHit.faceIndex]].name
TypeError: only integer scalar arrays can be converted to a scalar index
Error: Python script failed, check the message in the system console
```

Do you have any comments on this? I really appreciate your help in advance!

ln-12, 2022-08-06:
Could you please upload a minimal example as .blend file?

ozge-albayrak, 2022-12-03:
Sorry for extremely late response. I am using the script provided by use called script_usage.blend in the example_scenes folder. And when I run it I get this error.

tomhill206, 2024-01-03:
Do you remember if you got a solution to this issue? I have run into the same problem when trying to use the add-on with scripting.

tomhill206, 2024-01-03:
In case anyone else is confused by this: ticking the checkbox in the GUI is still necessary even if you are using the script-usage instructions.

ln-12, 2024-12-29:
I can reproduce the `only integer scalar arrays can be converted to a scalar index` error. It is obviously related to the sonar scanner but I could not figure out yet why it happens although it seems to be configured the same way as in the `sonar_example.blend` example.