ln-12/blainder-range-scanner #20: Rendered object surfaces in point cloud

awied1404, 2022-08-23:
Hi!
First of all thank you very much for sharing your code here. It will be very useful for me as I want to use it to create a synthetic dataset for a machine learning application.

I noticed that the points in the point cloud have the color of the material itself. What I would like to use are the colors of the rendered object surfaces, so like if using the ViewPort Shading mode.
I would like to know if it is possible or what I would have to do to get a point cloud that has these information.

Thank you for your help and best regards

ln-12, 2022-08-23:
Hey @awied1404, nice to hear that my tool is useful for you!

All available options for setting the object color are described [here](https://github.com/ln-12/blainder-range-scanner#diffuse-material). Unfortunately, my tool only supports `Princinpled BSDF` with a `Base Color` or texture at the moment. You need to set the desired color using this node. If you need any other shader nodes, pull request are highly welcomed :)

awied1404, 2022-08-24:
Okay it's a little bit clearer now, thank you!
As I stepped deeper in the code I saw that when using the castRay function in lidar.py the color of a point in the point cloud is determined. And I think that these functions don't consider the lightning when calculating the color of a certain point. Am I right with this assumption?

ln-12, 2022-08-25:
Yes that's right. The materials are mapped [here](https://github.com/ln-12/blainder-range-scanner/blob/7a9833279d0a5477e98311b23c066691a9f20ecd/range_scanner/material_helper.py#L31) and the color is retrieved [here](https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/material_helper.py#L98). Unfortunately, I don't see any simple possibilities to implement the support for shading as you would somehow have to get the rendered surface color from Blender.

awied1404, 2022-08-25:
Okay thank you for the confirmation. If I somehow would be able to get the rendered surface color, I will make a pr.

closed 2022-08-25 by awied1404