ln-12/blainder-range-scanner #21: ViewLayer does not contain object

thomasTUK, 2022-08-30:
Hello there,

I wanted to prepare different layers for sensor simulation, so I placed Collection.001 on one layer and Collection.002 in a different layer, see attached picture (still active in the View Layer)
[image]

No then I am executring the generation of point cloud for one scene, it does not work and I get the error that the other cube is not in the scene

[image]

Collection.001 and Collection.002 contain excactly the same. 
Is this a bug or meant to be? If meant to be, that's alright, just wanted to check in.

thomasTUK, 2022-08-30:
Here is the file 
[test.zip](https://github.com/ln-12/blainder-range-scanner/files/9451119/test.zip)

ln-12, 2022-08-31:
Hey @thomasTUK, I remember having issues with collections when implementing the add-on. I currently don't have time to look into it, but I will update you as soon as I know more.

thomasTUK, 2022-09-04:
Hi @ln-12  no worries, I did just split my Blender File into five each. So For now, I can do my simualtions :)

ln-12, 2022-10-03:
@thomasTUK commit https://github.com/ln-12/blainder-range-scanner/commit/840d48038697aee966eb48d0676679ff7d08abe7 fixes the issue for me. Could you please check it for your workflow?

closed 2022-10-03 by ln-12

thomasTUK, 2022-10-04:
Did work!

commit 840d4803 (2022-10-03): Getting object from view layer instead of scene
- fixes #21