ln-12/blainder-range-scanner #22: Update BlAINder for Version 3.3

thomasTUK, 2022-09-08:
"It is recommended to use Blender 2.93 LTS. The next LTS will be available with Blender 3.3 (see [here](https://developer.blender.org/project/view/142/)) for which the add-on will be updated. Support for that version is prepared in [this branch](https://github.com/ln-12/blainder-range-scanner/tree/Blender3). Feel free to open an issue if you face problems with Blender 3.x while using that branch.
WARNING: DO NOT install the addon via both ways or the two versions are mixed up and cause errors."

Is there already an updated version?

ln-12, 2022-09-08:
Yes, there is the https://github.com/ln-12/blainder-range-scanner/tree/Blender3 branch, which you can test with the latest version of Blender. When I have time, I will also check it myself and merge it into main.

ln-12, 2022-10-03:
Hey @thomasTUK, did you find any problem for your workflow so far?

thomasTUK, 2022-10-04:
Hej @ln-12 I did try, but did not work. I did not try again, since it is still working with my 3.0 version and I am not in a hurry to update.

ln-12, 2023-04-10:
Hey @thomasTUK, I just merged the Blender3 branch into main. I could not find any issues in my tests after installing Blender 3.3 LTS and the latest BlAInder version from scratch. If you still face any issue, please open a new issue.