ln-12/blainder-range-scanner #23: render error

zwqyfy, 2022-10-23:
i just open scence/sonar_example.blend
after click Render Animation,logs show
Render error (No such file or directory) cannot save: 'D:\Projekte\Masterarbeit\range_scanner\image_1.png0001.png'do nothing 
i am sure that set the export directory is the local directory
Is there a necessary step that has not been done?

ln-12, 2022-10-24:
Hey @zwqyfy, you are right, the rendering fails. This issue has nothing to do with my addon, but with the general Blender configuration. Please have a look at the following screenshot where you can see the path selector:

[image]

ln-12, 2022-10-24:
I fixed the issue in commit https://github.com/ln-12/blainder-range-scanner/commit/a692973c76e0745360289d9265d771fe9bff68b9 which uses the `//` placeholder instead of the local path.