ln-12/blainder-range-scanner #24: Ability for Channels

thomasTUK, 2022-11-28:
Hello, 
in my opinion the laser simulation is wrong. In the following screenshots you can see (or assume^^) that there are 122 (0,40° FOV devided by 0,33° angular resolution) points verically:

[image]

The Velodyne Ultra Puck has only 32 Channels (https://velodynelidar.com/wp-content/uploads/2019/12/63-9378_Rev-F_Ultra-Puck_Datasheet_Web.pdf), which means the points should also be 32 vertically. I got following mail from the velodyne support, since i was really wondering....

"In the Ultra puck sensor you have 32 laser beam distributed in a non linear manner in these 40 degrees vicinity giving you 0.33 degrees resolution in the middle line and it grows as you go toward the outer line."

Am i mistaken? 

Regards
Thomas

ln-12, 2022-12-01:
Well, you are probably right. When I wrote this plugin, I had the same data sheet but did not really know (or care) about the channel property. I think a quick workaround would be to set the vertical resolution to `1.25` as `40 / 1.25 = 32` to get an equally spaced distribution. Without a proper model of how the beams are actually distributed, I don't know how I should implement it. Or did they send any additional information about it?

thomasTUK, 2022-12-01:
I did do your workaround and it works well. I did ask about a proper model, but got the following response:

"No, you cannot calculate others degrees, its not changeable."

The horizontal laser line has a degree of 0,33°, but the others are non-linear distributed and you are right, without a proper model, there is nothing we can do.