ln-12/blainder-range-scanner #25: Error when trying to install dependencies (Linux)

joshm42, 2022-12-19:
**Bug Description:** When installing the range scanner add-on it tells me there are missing dependencies. Upon hitting the "Install dependencies" button I receive the following output in the console: 

"ERROR: Command '['/snap/blender/3082/3.4/python/bin/python3.10', '-m', 'pip', 'install', 'Jinja2==3.0.2']' returned non-zero exit status 1"

However, I can manually use that same python binary / pip to install Jinja2 and all of the other dependencies in the Readme. Despite being able to see them all listed in Blender's instance of pip (using pip list with the Blender python executable) the add-on still does not see them and still requires the "install dependencies" button to be pressed repeating the error.

When I run the subprocess command directly in Blender's console (command: subprocess.run([sys.executable, "-m", "pip","install", "Jinja2==3.0.2"],check=True) it does not error out and instead has "returncode=0". Despite this, the "missing dependencies" notice does not go away in Scanner tab. 

**To Reproduce:**
1) Snap install Blender (currently snap is installing Blender 3.4.0)
2) Launch Blender using the terminal
2) Download the Blainder repository
3) Extract all files and zip range_scanner folder
4) In Blender's add-on menu select the range_scanner zip folder 
5) Select "Scanner" from tab and press "Install dependencies" 
6) See output error in terminal

**Expected Behavior:**
I have used this add-on with Blender 2.93.9 on a Windows 7 machine and the add-on installs dependencies fine. I now need the developed pipeline to run on a Linux machine which is resulting in the error described above.

**Desktop:** 
-OS: Linux 20.04.5 
-Blender 3.4.0

ln-12, 2022-12-19:
I once tested the installation as a snap package, but it die not work for me as these packages cannot be altered after installation. Please use the method described in the readme instead.

ln-12, 2022-12-19:
Alright, thank you for testing it out. I will have a look at it, when I got some time.

ln-12, 2024-12-29:
I just updated the readme and checked it for Blender 4.2 on Windows. For me it works fine now, could you also try again?