ln-12/blainder-range-scanner #26: Csv file with random CategoryID and partID

HackerHakcer, 2023-03-10:
I am currently working with blender and i am trying to use a the BLAINDER add-on in order to simulated a lidar in a  scene and get the point cloud data in the csv file.
My problem is even when i assign a categoryID and a partID in the custom property to different objects and different parts of the same  object i still get some random number in the csv file that don't make any sense.
I want the data to be labeled in the csv file according to the custom property that i assign.
Any help will be appreciated

ln-12, 2023-03-20:
Could you please provide a .blend file with a minimal example? Otherwise I cannot reproduce it.

MojganFaramarzi, 2023-08-31:
Hi,
Thanks for the useful add-on you presented. 
I also have the same issue. It seems like it assigns indices to the objects and then extracts them as a label. I need the names of the objects instead of random indices. Is there any way that you help me edit the code to add the name of the object to the exported CSV file? I went through the code and tried to modify the 'data' passing through the 'export_csv.py', and 'scannedValues' in the 'generic.py'; but it is time-consuming to decode it. I would appreciate your help if possible :)

Thanks

ln-12, 2023-09-01:
Hey @MojganFaramarzi, I think you could expand the [hit info](https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/scanners/hit_info.py) according to your needs. You would just have to retrieve the object name and store it into the info object, for example [here](https://github.com/ln-12/blainder-range-scanner/blob/1131bb528998659180a5a5c1ea931051eaed6ba4/range_scanner/scanners/lidar.py#L424). Then you can map the data [here](https://github.com/ln-12/blainder-range-scanner/blob/1131bb528998659180a5a5c1ea931051eaed6ba4/range_scanner/export/exporter.py#L30) and export it [here](https://github.com/ln-12/blainder-range-scanner/blob/1131bb528998659180a5a5c1ea931051eaed6ba4/range_scanner/export/export_csv.py#L22).

Is that what you are looking for?

MojganFaramarzi, 2023-09-01:
Thanks a lot @ln-12  for the detailed reply. Yeah, I wanted to know which parts of the which files I should change.
I appreciate your time and help.

MojganFaramarzi, 2023-10-31:
> Could you please provide a .blend file with a minimal example? Otherwise I cannot reproduce it.

Hi again,
Can I ask if you realized where the issue is or have updated the code? My extracted file does not give partID and CategoryId that I define either, random numbers.

I appreciate your help

MojganFaramarzi, 2023-10-31:
I just tried to change the order of creating objects in blender. For my file, it seems like partID is assigned with the order that objects are created, and changing the order of two objects solved the problem for me :D 

Thanks

MojganFaramarzi, 2023-10-31:
It seems like the order of creating the objects in Blender matters. For my file, changing the place of the commands line to create the two intended groups, solved the issue :d

Thanks :)

closed 2023-11-25 by ln-12

MojganFaramarzi, 2023-12-04:
Sorry I have another question if you don't mind. Is there a straignforward way to extract the data in the sensor's coordinate frame? I think, in default, the data  is collected in the world coordinate frame.

Thanks

ln-12, 2023-12-04:
It is not part of the tool yet, but I think you could pass the camera properties to the ray casting function [here](https://github.com/ln-12/blainder-range-scanner/blob/1131bb528998659180a5a5c1ea931051eaed6ba4/range_scanner/scanners/lidar.py#L393) and apply your transformation to the `closestLocation` (the hit point) [here](https://github.com/ln-12/blainder-range-scanner/blob/1131bb528998659180a5a5c1ea931051eaed6ba4/range_scanner/scanners/generic.py#L175). 

Alternatively, you could export the camera properties and apply the transformation afterwards in an external tool.

A simple workaround might be to simply move your scene so that your camera is at the world origin.

MojganFaramarzi, 2023-12-06:
Thanks for your answer and suggestions.

Actually, I am doing the second one currently; However, I thought if there is a straightforward way to extract it directly from the tool, that might be better.

Since I want to use multiple scanners, I think I can not go with the last suggestion.
I appreciate your time and response.

oficinadeproyectoscs, 2024-11-11:
Hello, 

I am dealing with the same issue. I want to assign a categoryID to the 3D objects of the scene.

I am trying the method you propose [here](https://github.com/ln-12/blainder-range-scanner/tree/1131bb528998659180a5a5c1ea931051eaed6ba4?tab=readme-ov-file#classification) but it keeps assigning random values.

I also tried to change the code and add a new column in cvs file reference to the object name. But did not work either.
How did you fix @MojganFaramarzi ?

Thanks so much for any help.
Regards.