ln-12/blainder-range-scanner #28: How to export the 3D point cloud of the scene？

SuperEmperor000, 2023-03-21:
Hi, thank you very much for your work, it's great and very helpful for my research. I am having a problem and would like to get your help. Now that the plugin can export the scan image and depth map of the side scan sonar, I would like to know how to export the 3D point cloud map of this scene, which is the height information and plane coordinates of the scene.

ln-12, 2023-03-21:
Hey @SuperEmperor000, nice to hear that my project helps you!

If I understand your request correctly, your only option with the plgin would be to setup another scanner which renders the point cloud of the scene (e.g. a Kinect scanner).
 
However, I think that this does not really fit your use case. You probably have to look for another way to get such a height map from Blender with the included features. Wouldn't it be sufficient to export all points from your scene?

SuperEmperor000, 2023-03-22:
Now the exported point cloud is not complete, the real point cloud that is obscured by the shadow cannot be exported, I want to get the complete terrain information of the scene. I would like to know how to modify it based on your current code, I am new to Blender and not familiar with its features.

ln-12, 2023-04-10:
Well, if you need to have all pionts in the scene, I don't see any easy way to achieve this using my addon. The whole point of it is to simulate lidar scanners. However, it would be possible to implement such a behaviour on the base of Blainder but it would not be trivial as you would have to carefully rewrite the ray tracing part. 

I don't know a way to get the point cloud you want in Blender or any other addon yet, but if you find a solution for your problem please let me know.

closed 2023-04-10 by ln-12