ln-12/blainder-range-scanner #31: Export sensor poses (location and rotation)

shayan-nikoo, 2023-05-01:
Thanks for the nice tool. 
I create a curve for the camera animation and create many frames (Blender autokeying) along the curve for a smooth sensor movement (animation) and skip every 2 to 5 frames. 
Is it possible to export the sensor trajectory including the sensor poses (locations, rotations)? How can we do that?

ln-12, 2023-05-02:
Hey @shayan-nikoo, thanks for your request. I think your idea should be quite straight forward to implement. I would start by getting the camera parameters from Blender in each frame [here](https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/scanners/generic.py#L362). You could then write the needed values to a .csv file (or any other format). 

Another approach could be to extend the [`hit_info` class](https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/scanners/hit_info.py#L1) by adding some camera parameter properties, read the camera parameters [here](https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/scanners/lidar.py#L39) and write them in the output file for each ray (if that's what you need) in any of the exporters like shown [here](https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/export/export_csv.py#L22).

Does that help?

shayan-nikoo, 2023-05-03:
Thank you @ln-12, I don't need each ray but only the location and orientation in each frame, so the first solution should work. Do the below lines in the for loop do the job?  and then I can save it in a list and export it:
print(bpy.data.objects["Camera"].location)
print(bpy.data.objects["Camera"].rotation_euler)

In `lidar.py`, I don't find a loop for frames, where in the code I can get properties per frame? Or only `generic.py` is enough? I'm using the Velodyne sensor to simulate scanning.

closed 2023-05-08 by shayan-nikoo