ln-12/blainder-range-scanner #33: generate point cloud causes memory leak

dogiesalibi, 2023-06-17:
**range scanner addon cause memory leak on Mac os when I click 'generate point cloud' **

**To Reproduce**
Steps to reproduce the behavior:
1. open up the blender file inside.
2. Click on one of the scene cameras in the outliner, and go into camera view. 
3. Set it as scanning object. 
4. click generate point cloud botton on the scanner panel.

**Expected behavior**
It should generate a set of point. But Blender goes no-responsive for a while, and the memory usage keep increasing until 100GB and more, the system gives a warning about the system running out of app memory. need to force quit Blender.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: [Mac OS Monterey--12.3]
 - Blender version [3.3]

**Additional context**
It doesn't happen in all scenes, it tends to occur in larger scenes which contain more objects (but not sure about this correlation), I need to simulate e driving environment lidar scan. I tried to reduce the resolution or resolution scale, doesn't seems to help.

The file size is beyond the limit, I now share it with google drive. so here is the file, https://drive.google.com/file/d/1EW7QjIgdqLoUigjzD_oIHWHurSIZEJ3x/view?usp=sharing

ln-12, 2023-06-19:
Hey, thanks for the report. I am aware that the implemented approach is not ideal for large data sets. Unfortunately, I currently don't have the time to find a better solution.

However, if you find a fix for the issue, I would happily review any PRs.

Frenchman997, 2023-12-08:
I recently ran into the same issue when generating LiDAR scans of larger environments. When running the point cloud generation, blender would allocate more and more memory until the system ran out of memory and would be killed by the OOM reaper.
After some investigation, I found that the copying of Image texture arrays was causing the issue especially when working with large Image textures. A single 4k image texture caused memory consumption of more than 4 GB.
In addition the memory allocated by these arrays is not be freed if an exception is raised during the point cloud generation.
By switching to numpy arrays to store the image textures I was able to reduce the memory consumption by about an order of magnitued. The memory consumption could be reduced even further by switching to a lower width floating point numpy array such as float16 but that hasn't been necessary for my use case.
I created a PR that implements this fix: https://github.com/ln-12/blainder-range-scanner/pull/38

ln-12, 2023-12-09:
I just merged the PR. @dogiesalibi does this help with your issue?

binshaea, 2024-05-19:
Hi, I am facing the same issue but in my scene I am importing objects randomly in the scene and I tried to do the (scan_static()) in the script during animation. The blender somehow fill up large amount of memory then it crashes. The strange thing is that the same project is running smoothly in another laptop with same setup and same os with GPU enabled using the same driver. I would appreciate if anyone can guide on this issue.

dogiesalibi, 2024-05-25:
@ln-12 Hi, it does help, Thanks:) I'll close the issue.

closed 2024-05-25 by dogiesalibi