ln-12/blainder-range-scanner #34: return array of getTargetMaterials may include 'None' elements

YBachmann, 2023-08-16:
**Describe the bug**
If a material has a 'Material Output' node but there are no inpu nodes for the 'Material Output' node, the getTargetMaterials function in material_helper.py will return an array that contains a None value for this material. 
This is because in the function the value of 'links' will be an empty tuple and therefore in the loop 'for link in links:' no material will be set at the current targetMaterials[materialIndex] spot. 

Later on in the scanning process this causes the following exception: 
`
File "C:\Users\Yannic\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\range_scanner\material_helper.py", line 104, in getMaterialColorAndMetallic
    if material.texture is not None:
AttributeError: 'NoneType' object has no attribute 'texture'
`

**To Reproduce**
Steps to reproduce the behavior:
1. Create a material with only an 'Material Output' Node: 
[image]
2. Assign the material to an object and start a scan. 

**Expected behavior**
- Output an error message that tells the user that the material (and what material exactly) is faulty. 
- Optionally: Ignore faulty materials and continue scanning anyway. 

**Desktop (please complete the following information):**
 - OS: Windows
 - Blender version: 3.6.1

If you want I can create a fix for this and make a pull request. 
Adding a condition in the getTargetMaterials function to check if `links == ()` or 'len(links) == 0' should do the trick.

ln-12, 2023-08-23:
Thanks for reporting this. If you have the time, it would be great if you could submit a PR 👍🏻

YBachmann, 2023-08-23:
PR: https://github.com/ln-12/blainder-range-scanner/pull/35

closed 2023-08-31 by ln-12