ln-12/blainder-range-scanner #36: Ocean Modifier being Disabled

Enginuic, 2023-09-02:
I'm not sure if this is a bug or is this a requirement for the range scanner to function, I'm attempting to simulate a sonar scan that covers up to where the ocean surface meets structure (splash zone). For this I'd like to have a dynamic sea surface so I've created a plane with the ocean modifier, but every time I "generate point clouds" it seems the ocean modifier gets removed. and the scan only reflects a single frame from the distorted plane rather than a surface that changes frame by frame. Otherwise fantastic software!

Is this modifier being removed by design?

Thanks! 

[image]

Enginuic, 2023-09-03:
As a follow up I'm trying my best to understanding the python scripts as I'm new to blender but I see this block of code which removes the modifier, so it was obviously intentional. The comments are a little unclear, could you clarify the limitation in the scanner that makes it not react to a surface with a shape modifier? 

It does seem to work with an object that is scaled/moved over time, but if i comment this block, it leaves the modifier in place, but all 200 frames are scanned as if the ocean shape hasn't changed after frame 0. Am i correct in assuming that once the scan starts **an object CAN move/scale, but it's SHAPE is locked in**?

[image]

ln-12, 2023-09-03:
Hey @Enginuic, nice to hear that you like my addon! You are totally right in your comment edit, this piece of the script applies the modifier. When developing this part, I had the issue that I used different modifiers and got the original version of my geometry rather than the modified one (just as you described it). The easiest way for me was to apply the modifiers so that the geometry is actually changed in a way that I can use from the script. This was no problem, a I loaded preconfigured `.blend` files, ran the simulation and closed them without writing any state to it. So the next time I opened it, I had the exact same starting point.

As I have not worked with Blender and especially the scripting part for quite some time now, I don't have a tested solution for you. But I could imagine this working by running `modifier_apply` before each scan, perform the scan, undo the modifier and update the frame count iteratively. With a real quick search, I found [this](https://blender.stackexchange.com/questions/268787/how-does-the-undo-operation-work) and [this](https://blenderartists.org/t/undo-after-applying-modifier/629653) which might work.

When you figure out a working approach, please let me know :)

Enginuic, 2023-09-04:
So I spent WAY too much time trying to figure this out. Wrote a bunch of convoluted code to create and apply modifiers for every frame, just to have it produce the same result. I used "bpy.ops.wm.redraw_timer(type='DRAW_WIN_SWAP', iterations=1)" which updated the viewport on each iteration and could watch the water moving as it should but it would continue to draw rays one at a time in the shape of the first frame ignoring the water state.

At this point i knew it had to be something in the script "remembering" how one object was intersecting another. After lots of tracing through the code I came across  this comment which had me wonder if this was the issue. I commented it out and my code worked perfectly!

[image]

I suspected at this point all of my work was probably unnecessary and started over. All i needed to do was remove this so the world matrix would be recalculated and in the step where the modifiers were applied I simply had  it skip applying the modifier if it was of type "OCEAN" then everything else just worked out.

[image]

Now the beams are interacting with the waterline in a more realistic fashion. I'd say I'm frustrated, but I learned a lot in the last 24hrs about your code and blender scripting in general. Thanks

[image]

ln-12, 2023-11-25:
Nice to hear, that you could figure this out! So can we close this issue?

closed 2024-12-29 by ln-12