ln-12/blainder-range-scanner #37: How to increase contrast for depthmap?

kikc9130, 2023-10-03:
To whom this may concern,

I am currently exploring the use of this software to simulate a drone's depth image scan of a road with potholes in blender for a university project.

As such, I have modeled a simple meshgrid with craters in it to represent a road with pothole defects, and positioned the camera directly above it (to represent a top down drone's eye view of the road) like so:
[image]

I have applied the same material on the meshgrid object as the one applied to the original cube in the example file "script_usage.blend".

Using the function for static scans, I have been able to successfully scan the meshgrid and output the resulting depthmap as an image to a desired file location:
[image]

However, the problem is that, in keeping with real world dimensions, I have set the depth of the potholes to be quite shallow wrt. the scene dimensions, with depths ranging from 25 to 75mm. 
[image]

Therefore, there is a great lack of contrast in the resulting depth image:
[image]

I have verified that the LIDAR capabilities are working by rendering other objects (like the default arrangement in "script_usage.blend"). 

Is there any way that I can tweak the source code to increase the contrast/sensitivity of the scan so that the difference in depth, although small, could be more apparent? Such that it looks like the result of a blender Z-pass render (example shown below for a road with potholes of similar depth)?
[image]

I have tried looking around the source code for a bit but unfortunately it is beyond the scope of my undergraduate knowledge and problem solving ability.

Best regards

ln-12, 2023-10-22:
Hey, sorry for the late reply. You can play around with the depth values and the mapping here: https://github.com/ln-12/blainder-range-scanner/blob/main/range_scanner/export/export_depthmap.py#L10-L33

You just have to transform the interval to your needs and write the new values into the output file.

ln-12, 2023-11-25:
Do you have any more questions or can we close this issue?

kikc9130, 2023-11-29:
Dear ln-12,

Thank you for your answer, I had found another workaround to achieve a consistent depthmap gradient using blender's baseline z-pass render to complete the project. So yes I will close the issue with this comment.

Nonetheless, I appreciate your explanation and will no doubt go first to your amazing range scanner if faced with a similar project in the future. It is truly a marvelous software that you have designed.

Apologies for the late correspondence, I wish you all the best with any projects you may have.

Kind regards

closed 2023-11-29 by kikc9130