ln-12/blainder-range-scanner #41: Compatability issue with Blender 4.0

binshaea, 2024-03-25:
Hi,
I am trying to install Blainder for my project that was built using blender version 4.0. I am using Ubuntu and I tried to install the add on using terminal but I cannot see the add on and use it in my project. 
before that I tried to install it using GUI through add-on and I was able to add the scanner tab but when I click on it I see a massage asking to install dependencies. when I click on it I have this error :
Command '['/snap/blender/4300/4.0/python/bin/python3.10', '-m', 'pip', 'install', 'Jinja2==3.0.2']' returned non-zero exit status 1.
Is it mandatory for this to work that I use same blender version (3.3) and python 3.9?

I need your guidance on this please

binshaea, 2024-03-26:
Thanks @ln-12 for your reply. 

I uninstalled the snap version of blender 4.0 and then reinstalled it using the typical steps in the Readme file (for script usage section). installing dependencies using GUI is definitely not working and gives error. I installed them using terminal as mentioned. however I noticed that during installation we are asked to create a separate folder named (addons_contrib). with that I was not able to see the add on appears in the addon lists although the folder appears to be linked. so what I did is I moved the range_scanner folder to the original addon folder and restarted the blender and I was able to add range_scanner and it worked. Now I am able to add scanners and configure them but I am not sure if this wont cause any issues later-on when we start running simulation and collecting data.

Thanks for your helpful guidance.

binshaea, 2024-03-26:
Thanks @ln-12 for your reply. 

I uninstalled the snap version of blender 4.0 and then reinstalled it using the typical steps in the Readme file (for script usage section). installing dependencies using GUI is definitely not working and gives error. I installed them using terminal as mentioned. however I noticed that during installation we are asked to create a separate folder named (addons_contrib). with that I was not able to see the add on appears in the addon lists although the folder appears to be linked. so what I did is I moved the range_scanner folder to the original addon folder and restarted the blender and I was able to add range_scanner and it worked with me when I started blender from terminal. Now I am able to add scanners and configure them but only when I start new project from scratch. once I open my project in blender I am no more able to see the scanner tab available which is so weird. 
Also I am not sure if using blender 4 with this add on wont cause any issues later-on when we start running simulation and collecting data.

Thanks for your helpful guidance.

ln-12, 2024-12-29:
Hey @binshaea, sorry for the late reply. I just updated the readme for Blender 4.2 on https://github.com/ln-12/blainder-range-scanner/tree/blender_4.2_lts and successfully tested it on Windows. 

You are right that the addon handling was changed. The latest changes are described here: https://devtalk.blender.org/t/changes-to-add-on-and-themes-bundling-4-2-onwards/34593. When installing it via Blender's preference menu, the addon is persistent for me across restarts of Blender. Maybe you wanna check again.