ln-12/blainder-range-scanner #42: Animation - Modifiers are removed when starting simulation

han16nah, 2024-04-29:
**Description**
I want to scan an animated tree with Blender. The tree was created with Sapling Tree Gen as curve, then an armature is created and the curve is created to mesh and wind animation is added. Before clicking "generate point clouds", the animation works. However, my simulated point clouds do not show any animation effects and after the simulation, my tree mesh is suddenly static, i.e., the `windSway` and `Skin` modifiers are missing.

**To Reproduce**
Steps to reproduce the behavior:
1. Activate the sapling tree gen add-on (Edit -> Preferences -> Add-ons).
2. Add a tree: Type `Shift + A` -> Curve -> Sapling Tree Gen
3. In the GUI, change to "Settings: Armature" and activate "Use Armature" and "Make Mesh"
4. Change to "Settings: Animation" and activate "Armature Animation"
5. In the View Layer, collapse the "treeArm" object, select the "treemesh" object and assign a material. 
6. Set the camera to look at the tree.
7. Configure the settings in the Blainder Scanner Add-on (here: Generic lidar, rotating, etc.). Make sure to "Enable animation". 
8. Click "Generate point clouds"

**Expected behavior**
The generated point cloud clearly show that the tree was moving, i.e., the point clouds from the different frames are different. Furthermore, my animated scene object stays the way it is before the Blainder simulation (with the wind modifier, etc.)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Blender version: 3.3

ln-12, 2024-04-29:
Hey, I have not really an idea why the animation itself is not workong for you. For the calculations, the geometry needs to be final in a sense that modifiers/dynamic manipulations cannot be taken into calculation. Probably the way your trees are generated is not compatible.

The "missing modifiers" problem is expected, as all modifiers need to be applied prior to simulation, see: https://github.com/ln-12/blainder-range-scanner/blob/8ffcb8dbb4405d762a1a90ed10236fb8cceacac5/range_scanner/scanners/generic.py#L236

han16nah, 2024-04-29:
Ah okay, thank you for the quick reply. Then I will try to find out if I can modify the animation so it is compatible (maybe with keyframing?) or whether it is just not possible for my use case.

closed 2024-04-29 by han16nah