ln-12/blainder-range-scanner #45: Static scans

hpc100, 2024-09-11:
Hello,
Thank's for sharing your works.

I try to create synthetic rotating scans with following parameters : 
- horizontal field of view : 360 degrees
- vertical field of view : 300 degrees
My camera setup : (x=12.9, y=3.6, z=1.5 and rx = 90°, ry=0° and rz=90°)

But when i run your code, i don't have any blank spaces under the synthetic scanner (see linked image (top view of the scene)). Do you have any suggestions ?

[image]

Thank's in advance.

hpc100, 2024-09-11:
Here, an illustration of the wanted result (on other scene)
[image]

ln-12, 2024-12-29:
What value do you input for the vertical FOV parameter in the interface? Please keep in mind that 180° means full vertical coverage from bottom to top, so in your case I think you want to use 150°. Only for horizontal FOV, the maximum value is 360°.

hpc100, 2025-01-29:
Thank's for your feedback, i used 150° and it's works

closed 2025-01-29 by hpc100